Denkschmied/fire-and-ice #12: Generate list of sensors

Denkschmied, 2019-04-08:


spsmichi, 2019-04-08:
@Michi

StudentSchnitzel, 2019-04-09:
Servus Thomas, kannst du mir diesen task zuweisen? Selbst geht das nicht....

Typ | Modell | Funktion | Implementierung | Notizen
-- | -- | -- | -- | --
Drehzahlsensor |   | Raddrehzahl | Erfolgreich |  
Gyro | Adafruit2472 | Orientierung | TBC |  
Infrarot | SEN0206, aus   <https://www.dfrobot.com/product-1495.html> | Temperatur | TBC |  
Kamera |   | Bild | Erfolgreich |  
Lautsprecher |   |   | Erfolgreich | Aus   Arduino kit
LIDAR |   | Abstandsmessung | TBC |  
RGB LED Modul | KY-016 | Anzeige | TBC |  
Servo | SM-2309S | Bewegen | Erfolgreich |  
Ultraschallsensor | HC-SR04 | Abstandsmessung | Erfolgreich |

Denkschmied, 2019-04-11:
Kannst du die Tabelle ins Wiki verschieben?

StudentSchnitzel, 2019-04-12:
@Denkschmied 
Passt das so? https://github.com/Denkschmied/fire-and-ice/wiki/List-of-Sensors 
Ich konnte den wiki Eintrag leider nicht mit dem issue Verknüpfen

Denkschmied, 2019-04-12:
Schaut gut aus :-)

ampxtreme, 2019-05-02:
Welche der Sensoren sollten jetzt zwingend ins Fahrzeug integriert werden? Ich bräuchte die Information um die Konstruktion weiterzubringen. 
@mausausdruck1 / @StudentSchnitzel: Könnt ihr hier was sagen?

mausausdruck1, 2019-05-11:
2 x Speedsensor LM393 an den Achsen (oder etwas anderes, womit du sicher herausfindest, wo du bist)
mind. 1 x Ultraschall
Kamera
ggf RFID-Leser am Boden

ampxtreme, 2019-05-16:
+ Flame-Sensor für das Fire-Team, oder?

ampxtreme, 2019-05-20:
> Servus Thomas, kannst du mir diesen task zuweisen? Selbst geht das nicht....
> 
> Typ	Modell	Funktion	Implementierung	Notizen
> Drehzahlsensor	 	Raddrehzahl	Erfolgreich	Pflicht
> Flame Detection	Hyperion Arduino Compatible Flame Detection Sensor	Flame-Detection	Erfolgreich	Pflicht für “Fire-Team”
> Ultraschallsensor	HC-SR04	Abstandsmessung	Erfolgreich	Pflicht (1x mit Servo, oder 3x mit
> Servo	SM-2309S	Bewegen	Erfolgreich	Pflicht bei 1x Ultraschall
> Kamera	Raspberry Pi Camera Module	Bild à Objekterkennung?	Erfolgreich	Optional
> Gyro	Adafruit2472	Orientierung	TBC	Optional
> Infrarot	SEN0206, aushttps://www.dfrobot.com/product-1495.html	Temperatur	TBC	Optional
> Lautsprecher	 	 	Erfolgreich	Aus Arduino kit
> LIDAR	 	Abstandsmessung	TBC	 
> RGB LED Modul	KY-016	Anzeige	TBC	Optional

Habe obenstehende Tabelle geupdatet. Passt das so für @all? Dann wäre das der Beschluss-Stand. Bitte bis 22.05. zurückmelden, falls etwas nicht passt. Ansonsten wird das Ticket geschlossen.

mausausdruck1, 2019-05-21:
RFID für das Eis-Team.  Hat da jemand Daten dazu?

ampxtreme, 2019-05-29:
Alle Sensoren sind in der Liste vorhanden. Wird ins WIKI übertragen. Issue wird geschlossen.

closed 2019-05-29 by ampxtreme