Denkschmied/fire-and-ice #6: Configure Raspberry Pi for MATLAB

Denkschmied, 2019-04-08:
Create check list for preparing Raspberry Pi to be used with MATLAB

fabioaufinger, 2019-04-08:
@fabioaufinger

mrc-gkmn, 2019-04-08:
Assign me too if additional help ist needed. :)

ampxtreme, 2019-04-10:
Everyone should do that at home. It´s way faster than in the FH! It took me 5 Minutes for the configuration of the Pi via Matlab. There is also a good tutorial for the Raspi-Addin.

ampxtreme, 2019-04-30:
Has everybody done this? 
I have a little problem here. At home I can use the raspberry normal with matlab. But I can´t connect to any other network. 
Maybe someone could upload his working image of the Rasp. with working configuration? Then I can copy the data for everybody to their SD-Cards when needed.

fabioaufinger, 2019-04-30:
The image of my Raspberry has 16GB in size. I think an upload is not recommended, we should exchange it in the next session in Kufstein with an USB device.

ampxtreme, 2019-05-02:
Really? I think the SD Card has 16 gig. But not the Raspberry. But yes we can also do it in the FH - if not everybody needs it, cause the SD cards are very slow.

closed 2019-05-20 by ampxtreme