Denkschmied/fire-and-ice #8: Order sensors

Denkschmied, 2019-04-08:


nico45, 2019-04-09:
Mache ich.

bobby-oh, 2019-04-12:
Bettina kommentiert.

nico45, 2019-04-13:
[Ordering of Sensors - Fire and Ice.pdf](https://github.com/Denkschmied/fire-and-ice/files/3076610/Ordering.of.Sensors.-.Fire.and.Ice.pdf)

nico45, 2019-04-13:
Task erledigt. Werde die Datei auch zusätzlich in Wiki einfügen. 
Bei zusätzlich benötigten Ergänzungen bitte melden Thomas.

VG
Nico

nico45, 2019-04-13:
FYI @Denkschmied

Denkschmied, 2019-04-19:
Muss die Issue noch mal öffnen: Bei den Sensoren brauche ich bitte noch die Stückzahl. Bei den Anbietern am Besten auf Reichelt, Mouser oder Conrad ausweichen - Ebay und sonstige Hersteller sind leider nicht möglich (aus Abrechnungstechnischen Gründen).

nico45, 2019-04-22:
[Ordering of Sensors 2 - Fire and Ice.pdf](https://github.com/Denkschmied/fire-and-ice/files/3102993/Ordering.of.Sensors.2.-.Fire.and.Ice.pdf)

nico45, 2019-04-22:
Anbei das Angebot von den jeweils billigsten Anbieter der Sensoren. Habe Reichelt, Mouser und Conrad untereinander abgeglichen.
VG
Nico

nico45, 2019-04-22:
@Denkschmied bei benötigten Ergänzungen bitte wieder melden :)

closed 2019-04-22 by nico45