Denkschmied/fire-and-ice #9: Define test area for challenge

Denkschmied, 2019-04-08:


Eli93TRO, 2019-04-09:
Elias

mausausdruck1, 2019-04-25:
Ich erlaube mir, hier auch meinen Senf dazu zu gebe:

Es ist ja klar, dass die echten Szenarien eher als Aufhänger dienen als praktisch umsetzbar. Auch geht es ja hier eher um das Ziel, mehr oder weniger schwarmartikes Verhalten zu zeigen und eine Aufgabe zu lösen.

### Size -> Skalierung der Sensorik
Im Ernstfall wäre der Einsatz in großen Gebäuden, mit ggF Stockwerken oder auch größere Lawinenabgängen. Das heisst: >> 800 qm
Wenn wir das in der FH machen, müssten wir das skalieren.  Bei großen Gebäuden/Lawinenabgängen macht ein LIDAR oä sinn, wenn wir alles schrumpfen, gewinnt der LIDAR deutlich an Leistungsfähigkeit. Der "Fog of War" bzw. der Nebel des Unbekannten sollte mit skalieren um den Schwarm nicht unnötig zu machen. Deshalb schlage ich vor, nur Sonic/IR zu nutzen, so dass Mapping auch wirklich eine Schwarmaufgabe sein kann. 

### Target (Schwerpunkt Eis)
Das Target könnte irgendetwas sein, was per Kamera erkenbar ist (Rote Kiste). 

Insb. für das Eis-Team wäre es aber lustig, wenn es nicht direkt erkennbar ist, sondern etwas mehr Suchen benötigt. Hier schlage ich ein RFID-Aufkleber und 2-3 falsche Aufkleber auf dem Boden vor, so dass die Roboter darüber fahren müssen. Das hat den Vorteil, dass die Nutzung eines Suchmusters mehr Sinn ergibt.

### Die Rettung (Schwerpunkt Eis)
Wenn dem Vorschlag RFID gefolgt wird, sehe ich kein Schieben oder ähnliches. Aber man könnte z.B. durch ein Umkreisen des Ziels mit dem Robotern ein Ausgraben "darstellen" :)

TLDR: Lidar weglassen, damit das Mapping eine Aufgabe werden kann. RFID ermöglichen Suchmuster.