Denkschmied/fire-and-ice #20: 3D-Druck Files

nico45, 2019-05-11:


nico45, 2019-05-11:
|   |   |   |   |  
-- | -- | -- | -- | -- | --
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |

nico45, 2019-05-11:
Autor | Dateiname | Stückzahl | Höhe von   Layer | Infill   (in %) | Bemerkung
-- | -- | -- | -- | -- | --
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |

nico45, 2019-05-11:
Autor | Dateiname | Stückzahl | Layer   Height | Infill   (in %) | Bemerkung
-- | -- | -- | -- | -- | --
Nicolas Roth | Roboter Ultimo_1   Roboter Ultimo_2 | 2 | 0,2 | 10 | Bitte beide Dateien drucken 😊
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |

nico45, 2019-05-11:
Autor | Dateiname | Stückzahl | Layer Height | Infill (in %) | Bemerkung
-- | -- | -- | -- | -- | --
Nicolas Roth | Roboter Ultimo_1 | 1 | 0,2 | 10 | 17:26 h
Nicolas Roth | Roboter Ultimo_2 | 1 | 0,2 | 10 | 17:14 h
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |  
  |   |   |   |   |

nico45, 2019-05-11:
Autor | Dateiname | Stück- zahl | Layer   Height | Infill (in   %) | Dauer | Bemerkung | Status
-- | -- | -- | -- | -- | -- | -- | --
Nicolas   Roth | Roboter   Ultimo_1 | 1 | 0,2 | 10 | 17:26   h |   |  
Nicolas   Roth | Roboter   Ultimo_2 | 1 | 0,2 | 10 | 17:14   h |   |  
  |   |   |   |   |   |   |  
  |   |   |   |   |   |   |  
  |   |   |   |   |   |   |  
  |   |   |   |   |   |   |  
  |   |   |   |   |   |   |  
  |   |   |   |   |   |   |  
  |   |   |   |   |   |   |  
  |   |   |   |   |   |   |  
  |   |   |   |   |   |   |  
  |   |   |   |   |   |   |  
  |   |   |   |   |   |   |  
  |   |   |   |   |   |   |  
  |   |   |   |   |   |   |  
  |   |   |   |   |   |   |  
  |   |   |   |   |   |   |  
  |   |   |   |   |   |   |  
  |   |   |   |   |   |   |  
  |   |   |   |   |   |   |  
  |   |   |   |   |   |   |  
  |   |   |   |   |   |   |  
  |   |   |   |   |   |   |  
  |   |   |   |   |   |   |

Denkschmied, 2019-05-11:
FH-Kufstein - Ultimkaer S5
Wattens - Ultimaker 3

ampxtreme, 2019-05-14:
Sorry, ich habe das hier erst jetzt gesehen - hatte die Files per Mail gesendet. Ich werde das heute Abend zuhause direkt hier hochladen.

ampxtreme, 2019-05-14:
> Sorry, ich habe das hier erst jetzt gesehen - hatte die Files per Mail gesendet. Ich werde das heute Abend zuhause direkt hier hochladen. Muss hier noch etwas beachtet werden beim Erstellen der Cura-Dateien?

Ich habe hier jetzt .ufp Dateien in Cura ausgespuckt bekommen. Passt das so auch @Denkschmied ?

mausausdruck1, 2019-05-15:
Irgendwer hat mich aus der Liste gelöscht. Für alle Fälle:

falk | falks_iceBot - stossdaempferhs-1.STL | 1 | egal | ? |  |  gerne flexibel |

Denkschmied, 2019-05-15:
bitte mit stl arbeiten - nur bei speziellen files die fertigen ufp verwenden

ampxtreme, 2019-05-16:
Meine Teile müssen im richtigen Winkel auf die Druckplatte gesetzt werden, sonst passt das nicht drauf. Ich lade aber auf jeden Fall die STL-Files hoch. 
Die .ufp Files sind dann nur falls es anders nicht klappt.

JoeUniv, 2019-05-24:
Von mir fehlen noch zwei Teile in der Liste:
- Blaulicht
- Körper
oder sind die schon gedruckt?

Denkschmied, 2019-05-24:
> Von mir fehlen noch zwei Teile in der Liste:
> 
> * Blaulicht
> * Körper
> 
> oder sind die schon gedruckt? @Denkschmied

Bitte Files hochladen (wenn nicht vorhanden) und in der Lsite ergänzen

fabioaufinger, 2019-05-24:
@Denkschmied muss ich die überarbeiteten Teile, die ich dir heute geschickt habe und die erneut zum Drucken wären, auch nochmal hier eintragen?

JoeUniv, 2019-05-26:
@Denkschmied Habe dir die Teile nochmal per Mail zukommen lassen.