Denkschmied/fire-and-ice #22: Bill of materials erstellen

Denkschmied, 2019-05-11:


JoeUniv, 2019-05-11:
**Kugelrollen (E37.019)**
Steel Ball: mit einem Durchmesser von 19mm 
**Link:** https://de.misumi-ec.com/vona2/detail/221006327744/?KWSearch=kugelrolle&searchFlow=results2products

**Preis** 7.14€

fabioaufinger, 2019-05-11:
|Anzahl|Komponente|Beschreibung|Link|
|:---|:---|:---|:---|
| ?  |Kugelrolle|als Stützrad für 2-Radantriebskonzepte| https://de.misumi-ec.com/vona2/detail/221006327744/?KWSearch=kugelrolle&searchFlow=results2products  |

fabioaufinger, 2019-05-14:
Die Kugelrolle ohne Einbautiefe könnte für viele Konzepte zu hoch sein bzw. nicht in die Bodenfreiheit passen. Bitte kontrollieren

mausausdruck1, 2019-05-16:
Meine Bodenplatte ist schon fertig, darum passt das gut in mein Konzept

ampxtreme, 2019-05-20:
Sind ausreichend Akkus vorhanden @Denkschmied oder sollen wir hier noch welche auf die Stückliste setzen?

Denkschmied, 2019-05-22:
Akkus von der Taschenlampe haben wir genügend.

daholzmichl, 2019-05-23:
Wir benötigen zudem H-Brücken sowie Geschwindigkeitssenoren.

https://www.amazon.de/gp/product/B07DK6Q8F9/ref=crt_ewc_title_dp_3?ie=UTF8&psc=1&smid=A3SCFTIO8CSK1X

https://www.amazon.de/LM393-Chip-Measuring-Komparator-Sensor-Modul-Optokoppler/dp/B0776RHKB1/ref=sr_1_3?keywords=speed+sensor&qid=1558640011&s=computers&sr=1-3

fabioaufinger, 2019-05-24:
Materialliste bitte jeder selber bearbeiten und seine Teile eintragen. Danke

mrc-gkmn, 2019-05-24:
|Anzahl|Komponente|Beschreibung|Link|
|:---|:---|:---|:---|
|1 | Kugelrolle|ohne Einbautiefe |https://www.conrad.de/de/p/kugelrolle-gehaeuse-stahl-verzinkt-2-loch-befestigung-typ-kr-f-25-800032286.html |
|1 | Wippschalter| Hauptschalter | https://www.conrad.de/de/p/wippschalter-a12131121000-250-v-ac-10-a-1-x-aus-ein-rastend-1-st-701661.html |
|1 | Drehzahlsensor| Hauptschalter | https://www.conrad.de/de/p/arduino-erweiterungs-platine-lm393-1646891.html |

mrc-gkmn, 2019-05-24:
> 
> 
> Materialliste bitte jeder selber bearbeiten und seine Teile eintragen. Danke

@fabioaufinger  Soll ich das gleich oben einfügen?

fabioaufinger, 2019-05-24:
> > Materialliste bitte jeder selber bearbeiten und seine Teile eintragen. Danke
> 
> @fabioaufinger Soll ich das gleich oben einfügen?

ja bitte :)

JoeUniv, 2019-05-24:
|Anzahl|Komponente|Beschreibung|Link|
|:---|:---|:---|:---|
| 1 |Kugelrolle 19 mm, Nylonball|als Stützrad für 2-Radantriebskonzepte| https://de.misumi-ec.com/vona2/detail/221006327744/?KWSearch=kugelrolle&searchFlow=results2products  |
|1 | Wippschalter| Hauptschalter | https://www.conrad.de/de/p/wippschalter-a12131121000-250-v-ac-10-a-1-x-aus-ein-rastend-1-st-701661.html |
|2 | Drehzahlsensor| Lichtschranke mit Lochscheibe zum Aufstecken auf DC Motor| https://www.conrad.de/de/p/arduino-erweiterungs-platine-lm393-1646891.html |

mausausdruck1, 2019-05-24:
Was ist denn jetzt die richtige Liste?

fabioaufinger, 2019-05-24:
> Was ist denn jetzt die richtige Liste?

Bei einer Liste ist die Auswahl nicht so groß würde ich sagen ;)

ampxtreme, 2019-05-24:
> > Was ist denn jetzt die richtige Liste?
> 
> Bei einer Liste ist die Auswahl nicht so groß würde ich sagen ;)

Im Wiki ist eine und  hier auch.

fabioaufinger, 2019-05-24:
> Im Wiki ist eine und hier auch.

Ich führe die Listen hier zusammen. Hier kann es jeder einfacher bearbeiten.

Eli93TRO, 2019-05-25:
1 | Wippschalter | Hauptschalter | https://www.conrad.de/de/p/tru-components-wippschalter-r13-112d-b-b-pr-i-o-ii-250-v-ac-6-a-1-x-ein-aus-ein-rastend-0-rastend-1-st-1565956.html
-- | -- | -- | --
1 | Taster | Reset/Start Taster | https://www.conrad.de/de/p/tru-components-tc-r13-527a-02bk-drucktaster-250-v-ac-6-a-1-x-aus-ein-tastend-1-st-1587757.html
1 | Kugelrolle | ohne Einbautiefe | https://www.conrad.de/de/p/kugelrolle-gehaeuse-stahl-verzinkt-2-loch-befestigung-typ-kr-f-25-800032286.html
2 | Drehzahlsensor | Lichtschranke mit   Lochscheibe zum Aufstecken auf DC Motor | https://www.conrad.de/de/p/arduino-erweiterungs-platine-lm393-1646891.html
2 | H-Brücken | zur Motorsteuerung | https://www.conrad.at/de/p/entwickler-platine-sbc-motodriver2-arduino-banana-pi-cubieboard-pcduino-raspberry-pi-raspberry-pi-2-b-raspberry-1573541.html
20 | CSPPNH01B-STN-M1.4-4 | Gewindeformende   Schrauben d1.4 | Misumi
1 | Servo_Motor | SM-2309S |  
1 | Camera | Bild |

fabioaufinger, 2019-05-28:
Bitte jeder die Liste oben noch einmal überprüfen und ggf. noch Teile eintragen.

Denkschmied, 2019-06-04:
Bestellungen wurden getätigt - sollte alles bis Freitag / Samstag da sein; bei den Kugelrollen gibt es eine Änderung (siehe zusätzliche Spalte in der Tabelle); Abmessungen sollten halbwegs den ursprünglichen Teilen entsprechen; Auswahl aufgrund Lieferzeit

ampxtreme, 2019-06-04:
@Denkschmied: Danke für die Info. Kannst Du uns noch Infos geben zum Ablauf am Freitag / Samstag?

daholzmichl, 2019-06-04:
Ich brauch nix, Danke!