Denkschmied/fire-and-ice #23: 3D-Druck Teil Unterbau_Deckel bitte checken

Denkschmied, 2019-05-14:
stl file besteht aus 2 objekten; eines davon ist so nicht druckbar (overhang) - bitte splitten

fabioaufinger, 2019-05-14:
Habe es extra so aufgeteilt wie wir das das letzte Mal ausgemacht haben. Ich schaue es mir am Abend an, vielleicht ist mir ein Fehler unterlaufen.

Denkschmied, 2019-05-14:
[image]
In gelb markiert der Überstand - wenn ich von rechts nach links drucke (rechts unterste schicht), dann geht sich die nase nicht aus

fabioaufinger, 2019-05-14:
Achso das ist nicht das Teil Unterbau_Deckel sondern Motorhalter. Wäre so gedacht gewesen, dass man die an der Kante aufstellt und die "Nase" mit Stützmaterial druckt. Bei mir im Cura ging es so.

Denkschmied, 2019-05-14:
Schickst du mir bitte einen Screenshoot in Cura?

closed 2019-05-20 by ampxtreme