Denkschmied/fire-and-ice #24: Treiber-Board Spec

mausausdruck1, 2019-05-21:
Wird ein neues Board entworfen, oder das aus Projekt 1 oder 2 verwendet? Wenn ersteres der Fall ist, wie sieht der Zeitplan aus? @Denkschmied

Denkschmied, 2019-05-22:
Wir werden ein fertiges Board verwenden. Im Stile von https://www.reichelt.at/entwicklerboards-motodriver2-debo-motodriver2-p202829.html?PROVID=2788&gclid=Cj0KCQjww47nBRDlARIsAEJ34bnXL7yJTSwFsslczNWwKGQBRuRN06fiEkK6nVZUpP4TCrIKLb4wXPMaAmvhEALw_wcB&&r=1

Denkschmied, 2019-05-22:
Hier kommt das finale Board:
https://www.conrad.at/de/p/entwickler-platine-sbc-motodriver2-arduino-banana-pi-cubieboard-pcduino-raspberry-pi-raspberry-pi-2-b-raspberry-1573541.html?ef_id=Cj0KCQjww47nBRDlARIsAEJ34bmS0T-gL1rZc5p9STetRc-nmbEg_X-jB6_hybyf-xu9t9la2r0aAwwaAj6iEALw_wcB:G:s&gclid=Cj0KCQjww47nBRDlARIsAEJ34bmS0T-gL1rZc5p9STetRc-nmbEg_X-jB6_hybyf-xu9t9la2r0aAwwaAj6iEALw_wcB

closed 2019-05-22 by mausausdruck1