Denkschmied/fire-and-ice #25: |Anzahl|Komponente|Beschreibung|Link|

JoeUniv, 2019-05-24:
|Anzahl|Komponente|Beschreibung|Link|
|:---|:---|:---|:---|
| 3 |Kugelrolle 19 mm, Nylonball|als Stützrad für 2-Radantriebskonzepte| https://de.misumi-ec.com/vona2/detail/221006327744/?KWSearch=kugelrolle&searchFlow=results2products  |
|1 |Wippschalter |Hauptschalter  | https://www.conrad.de/de/p/tru-components-wippschalter-r13-112d-b-b-pr-i-o-ii-250-v-ac-6-a-1-x-ein-aus-ein-rastend-0-rastend-1-st-1565956.html |
|1 | Taster| Reset/Start Taster| https://www.conrad.de/de/p/tru-components-tc-r13-527a-02bk-drucktaster-250-v-ac-6-a-1-x-aus-ein-tastend-1-st-1587757.html|
|1 | Kugelrolle|ohne Einbautiefe |https://www.conrad.de/de/p/kugelrolle-gehaeuse-stahl-verzinkt-2-loch-befestigung-typ-kr-f-25-800032286.html
|2 | Wippschalter| Hauptschalter | https://www.conrad.de/de/p/wippschalter-a12131121000-250-v-ac-10-a-1-x-aus-ein-rastend-1-st-701661.html |
|4 | Drehzahlsensor| Lichtschranke mit Lochscheibe zum Aufstecken auf DC Motor| https://www.conrad.de/de/p/arduino-erweiterungs-platine-lm393-1646891.html |

_Originally posted by @fabioaufinger in https://github.com/Denkschmied/fire-and-ice/issues/22#issuecomment-491513788_

JoeUniv, 2019-05-24:
Aaah sorry

fabioaufinger, 2019-05-24:
Listen wurden zusammengeführt.

closed 2019-05-24 by fabioaufinger