Denkschmied/fire-and-ice #30: Python Code for RPM-Sensor

Denkschmied, 2019-05-24:
**Ziel:**
Funktionierendes Programm welches die Umdrehung eines Rads wiedergibt. Ausgegeben werden soll die Umdrehung in 1/min.

Folgender Sensor wird verwendet werden:
http://anleitung.joy-it.net/?goods=speed-sensor

Denkschmied, 2019-05-24:
https://github.com/Denkschmied/fire-and-ice/wiki/Software

ampxtreme, 2019-05-29:
Fahrgeschwindigkeit | 5 km/h
-- | --
Fahrgeschwindigkeit | 1,39 m/s
Durchmesser Rad: 6 cm | 0,19 m
20 Löcher/U (Lochscheibe) | 7,37 U/s
Abtastfrequenz | 147 Hz
Nyquist | 295 Hz
Sicherheitsfaktor (2x RPM) | 590 Hz
Durchlaufzeit | 2 ms

ampxtreme, 2019-06-03:
Habe den Code nun hochgeladen. Hat jemand einen RPM und kann ihn mal ausprobieren? 
@daholzmichl : Du villeicht?

closed 2019-06-04 by ampxtreme