Denkschmied/fire-and-ice #31: Python Code for LIS3DH

Denkschmied, 2019-05-24:
**Ziel:**
Funktionierendes Programm für den Sensor LIS3DH (Orientierungssensor). Das Programm soll die Orientierung des Fahrzeugs (als Kompassfunktion) ausgeben.

Details zum Sensor:
https://learn.adafruit.com/adafruit-lis3dh-triple-axis-accelerometer-breakout/overview

Denkschmied, 2019-05-24:
https://github.com/Denkschmied/fire-and-ice/wiki/Software

mrc-gkmn, 2019-05-24:
Würd ich übernehmen und versuchen einen funktionierenden Code zu schreiben 👍

mrc-gkmn, 2019-05-26:
### Hardware Aufbau 
[image]

| Raspberry Pi  |  MPU 6050 bzw. GY-521 (beide identisch)  |
|---|---|
| PIN 1 (3.3V) | VCC  |
| PIN 3 (SDA)  | SDA  |
| PIN 5 SCL | SCL  |
| PIN 6 (GND)| GND  |

### I²C aktivieren
Als erstes muss man SPI & I2C aktivieren. Dieser Schritt kann übersprungen werden, wenn dies bereits geschehen ist.
Dazu `sudo raspi-config`. Unter '5. Interfacing Options' gibt es die Einträge "P4 SPI" und "P5 I2C". Diese müssen enabled werden.
Anschließend modules-Datei mit `sudo nano /etc/modules` bearbeiten und folgende Zeilen zur Datei hinzufügen (wenn nicht bereits vorhanden): 
`i2c-bcm2708
i2c-dev`
Anschließend per Strg+O speichern und per Strg+X beenden.

Jetzt die nötigen Tools installieren:
`sudo apt-get install i2c-tools python-smbus`
Testen des Tools mit:
`sudo i2cdetect -y 1` - Hier sollte bei richtig angeschlossenem Gyro-Modul eine Zahl anstelle der "--" stehen. 
Beispiel: 
> pi@raspberrypi ~ $ sudo i2cdetect -y 1
>  0 1 2 3 4 5 6 7 8 9 a b c d e f
> 00: -- -- -- -- -- -- -- -- -- -- -- -- --
> 10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
> 20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
> 30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
> 40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
> 50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
> 60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- --
> 70: -- -- -- -- -- -- -- --
Hier befindet sich das Gerät auf Adresse 68 (Hexadezimal). Um das Register anzusprechen folgendes eingeben: `sudo i2cget -y 1 0x68 0x75`

### Python Script
[gyro.txt](https://github.com/Denkschmied/fire-and-ice/files/3220104/gyro.txt)
Datei muss im .py-Format gespeichert werden (upload der .py-Datei war nicht möglich daher .txt) 

Um das Skript erfolgreich starten zu können muss in line 37 die Adresse auf den Wert der i2c-Abfrage gesetzt werden.  

Hier eine Übersicht der wichtigsten Adressen auf dem Gyro zur Ausgabe verschiedener Werte:
- PWR_MGMT_1   = 0x6B
- SMPLRT_DIV   = 0x19
- CONFIG       = 0x1A
- GYRO_CONFIG  = 0x1B
- INT_ENABLE   = 0x38
- ACCEL_XOUT_H = 0x3B
- ACCEL_YOUT_H = 0x3D
- ACCEL_ZOUT_H = 0x3F
- GYRO_XOUT_H  = 0x43
- GYRO_YOUT_H  = 0x45
- GYRO_ZOUT_H  = 0x47

Diese wurden größtenteils im Skript (gyro.py) verwendet.

Beispiel der Ausgabe des Skriptes:
> Gyroskop
> --------
> gyroskop_xout:   -260  skaliert:  -2
> gyroskop_yout:   -154  skaliert:  -2
> gyroskop_zout:     78  skaliert:  0
> 
> Beschleunigungssensor
> ---------------------
> beschleunigung_xout:   -1048  skaliert:  -0.06396484375
> beschleunigung_yout:    -676  skaliert:  -0.041259765625
> beschleunigung_zout:   16644  skaliert:  1.01586914062
> X Rotation:  -2.32121150537
> Y Rotation:  3.59994842011

Skalierte Werte werden hierbei dazu verwendet, die Ungenauigkeiten der Werte zu entschärfen und die Wertebereiche eindeutiger und leichter interpretierbar zu formen. Speziell für unsere Anwendung sind daher die Skallierten Werte brauchbarer als die RAW-Values des Sensors selbst.

Um eine alternative Herangehensweise aufzuzeigen hier noch ein zweites Skript:
[gyro alternative.txt](https://github.com/Denkschmied/fire-and-ice/files/3220132/gyro.alternative.txt)

### Hilfreiche Unterlagen
- https://tutorials-raspberrypi.de/rotation-und-beschleunigung-mit-dem-raspberry-pi-messen/

Zur Messung des Neigungswinkel zwischen zwei Achsen empfehle ich noch diesen Artikel:
- http://www.hobbytronics.co.uk/accelerometer-info

Mehr Unterlagen:
- https://www.electronicwings.com/raspberry-pi/mpu6050-accelerometergyroscope-interfacing-with-raspberry-pi
- https://www.youtube.com/watch?v=wTfSfhjhAU0&app=desktop
- http://www.mschoeffler.de/2017/10/05/tutorial-how-to-use-the-gy-521-module-mpu-6050-breakout-board-with-the-arduino-uno/

+++ leider konnte ich die Scripte wegen fehlender Hardware nicht testen, bin sie aber sehr detailliert durchgegangen und bin überzeugt dass diese funktionieren, wenn man dieser Anleitung folgt +++

mrc-gkmn, 2019-05-26:
Hilfreicher Tipp zur Realisierung ist auch die Rotationsrichtung der x- bzw y- Achse. Diese ist wie auf folgender Grafik zu erkennen, auf der unterseite der Platine eingezeichnet. 
[image]

ampxtreme, 2019-06-01:
Habe den Code mal als .py File ins Repository gestellt