Denkschmied/fire-and-ice #35: Concept of random robot movement

Denkschmied, 2019-05-24:
**Ziel:**
Einfacher Algorithmus für das Bewegungsmuster der Roboter. Einbindung einer Zufallsvariabel um random movement zu erzeugen.

Denkschmied, 2019-05-24:
Aufbauend auf Konzepte soll hier ein Algorithmus entstehen, wie sich die Fahrzeuge im Raum bewegen sollen. Ziel ist es, dass jedes Fahrzeug einen anderen Pfad einschlät und so das Terrain erkundigt wird. Teil des Algorithmus ist auch die Vorgabe, wie die Roboter sich bewegen sollen. Im Sinne von:
- Fahren
- Stop
- Datenerfassung
- Datenübertragung
- Fahren

JoeUniv, 2019-05-27:
https://www.youtube.com/watch?v=4BdY8MPud48

ampxtreme, 2019-05-27:
Ich habe gerade nur in das Video rein gesehen. Schaut sehr interessant aus. Ist das was für uns als Ausgangsbasis?