Denkschmied/fire-and-ice #36: Pyhton Code for Camera - Image Acquisiton

Denkschmied, 2019-05-24:
**Ziel:**
Bildauswertung am Raspberry Pi erkennt, um welches Objekt es sich bei einem Hindernis handelt. Aufbauend auf die Arbeiten im Wiki zum Thema Kamera soll eine funktionierende Software entstehen, welche zuverlässig vordefinierte Objekte auf dem Bild erkennt.

fabioaufinger, 2019-05-24:
Hab in der Wiki bei den Code Snippets unseren MarsRover Code für die Kamera zur Verfügung gestellt.

Denkschmied, 2019-05-27:
Hmm...gute Frage. Wir können den zu suchenden Gegenstand vorab definieren. Wie wäre es, wenn wir die roten Hindernisse aus der MarsRover Challenge recyclen? Oder ein farbiges Quadrat mit definierter Größe? Oder Kreis?

ampxtreme, 2019-05-27:
> Hab in der Wiki unseren MarsRover Code für die Kamera zur Verfügung gestellt. Dieser ist im Moment nur zur Farberkennung und gibt die Objektkoordinaten im Bild aus.
> @Denkschmied inwiefern soll eine Objekterkennung durchgeführt werden? Sollen verschiedene Objekte eintrainiert werden oder gehen wir nach anderen Kriterien vor?

Die Installation wird dann aber nicht auf jedem Roboter vorgenommen, oder? So wie ich das verstanden habe ist das Setup sehr aufwändig. Außer wir gehen auf ein Image was sich alle auf die SD-Karte ziehen können vorab...

fabioaufinger, 2019-05-28:
Wir haben noch das zusätzliche Problem, dass bei dem vorhandenen Image OpenCV in einem virtual environment installiert ist. Bis heute haben wir es nicht geschafft dieses über diverse Startskripte etc. anzusprechen. Somit ist eine autonome Lösung nicht möglich (jeder müsste die Roboter über Konsolenbefehle starten).
Ich versuche morgen zu Hause OpenCV auf einem Image ohne virtual environment zu installieren, inkl. aller zusätzlichen Bibliotheken für das Projekt. Dieses Image können wir dann klonen (@ampxtreme ) und jeder kann es verwenden.
@Denkschmied : vl brauchen wir hierzu noch SD-Karten für jeden, da die vorhandenen schreibgeschützt bzw. kaputt sind.

fabioaufinger, 2019-05-28:
> Hmm...gute Frage. Wir können den zu suchenden Gegenstand vorab definieren. Wie wäre es, wenn wir die roten Hindernisse aus der MarsRover Challenge recyclen? Oder ein farbiges Quadrat mit definierter Größe? Oder Kreis?

Ich glaube die roten Kisten aus der MarsRover Challenge sind am einfachsten.

mausausdruck1, 2019-06-05:
Vor allem sehen alle Bots anders aus. Wenn wir noch die Wand/Bäume sicher erkennen, müssten wir nicht über statistische Methoden schließen, wo wahrscheinlich ein Bot ist.