Denkschmied/fire-and-ice #38: Matlab und Python, was ist die Ziel-Architektur?

mausausdruck1, 2019-05-28:
@Denkschmied: bisher gehen wir davon aus, dass sich aller Python Code in Matlab portieren lässt und die Zielarchitektur Matlab ist, oder?

Die Annahme sollte sorgfältig geprüft werden. Mir fällt dazu ein: 
1) der Bild- CV-code vom Rover basiert auf Python. Eine Portierung oder neu-Implementierung muss geprüft werden.
2) Interrupts lassen sich in Python umsetzen. Die Matlab-Bibliothek unterstützt das nicht. Ohne Interrupts können wir die Speedsensoren nicht sicher direkt auslesen (Loop-Problem!). Da bleibt dann nur ein Workaround via Arduino als externer Zähler, der Interrupts kann.

Denkschmied, 2019-06-03:
Folgender Vorschlag:
- Matlab für die Visualisierung der Karte
- Steuerung der Roboter über Python

Denkschmied, 2019-06-03:
Wo stehen wir denn mit den Code Snippets? Soweit ich jetzt gelesen habe, gibts bei MATLAB Probleme. Können diese in Python behoben werden?

MartinSchafferer, 2019-06-04:
Bekomm immer wieder die Fehlermeldung, dass User oder Passwort nicht stimmen, wenn ich versuche ein File vom RPi zu Matlab zu schicken bzw. umgekehrt. Hab den Fehler noch nicht gefunden und ja Passwort und User sind richtig ;)

ampxtreme, 2019-06-04:
@MartinSchafferer: Wenn Du den Code zur Verfügung stellt, kann ich es auch mal probieren.

mausausdruck1, 2019-06-05:
Ich fasse zusammen: Roboter erst einmal bevorzugt über Python steuern und messen lassen. Über die Kommunikation später gedanken machen. (Die Matlab-Programme, die wir bisher getestet haben, sind nämlich ein anderer Loop. Lässt sich sicher irgendwie lösen, Multiloop, MQTT oder so ).

Nachteil: Roboter haben keine digitalen Matlab-Zwilinge.

Richtig?

Denkschmied, 2019-06-21:
https://de.mathworks.com/products/matlab/matlab-and-python.html?s_eid=psm_ml&source=15308.html

Vielleicht hilft uns das.