Denkschmied/fire-and-ice #39: Übersicht Loop-Durchlaufzeiten (Fehlende Interrupts bei MATLAB)

ampxtreme, 2019-05-29:
Bezeichnung | Wert
-- | --
Fahrgeschwindigkeit | 5 km/h
Fahrgeschwindigkeit | 1,39 m/s
Durchmesser Rad: 6 cm | 0,19 m
20 Löcher/U (Lochscheibe) | 7,37 U/s
Abtastfrequenz | 147 Hz
Nyquist | 295 Hz
Sicherheitsfaktor (2x RPM) | 590 Hz
Durchlaufzeit

ampxtreme, 2019-05-29:
Sensor | Durchlaufzeit [ms]
-- | --
RPM | 2
Ultraschall |  
Servo + Ultraschall |  
Camera |  
Gyro |  
Flame |

ampxtreme, 2019-05-29:
Falls die Durchlaufzeiten zu hoch werden sollten, können wir folgende Ansätze verfolgen:

- Inner- und Outer-Loops, Innerhalb des Inner-Loops würden die Sensoren und Fahren laufen. Im Outer-Loop der Rest. Beides gleichzeitig geht in diesem Fall dann nicht. (Beispiel: https://repository.asu.edu/attachments/162093/content/Lin_asu_0010N_15358.pdf)
- Zweiten Microcontroller implementieren und darüber die Sensoren auslesen. Komm. über RS232/UART (ggf. Bauraumprobleme)

- Siehe auch: https://github.com/Denkschmied/fire-and-ice/issues/38

fabioaufinger, 2019-05-29:
Hab noch ein Ass im Ärmel. Eine weitere Möglichkeit wäre über ein Bash-Skript zwei Python-Programme gleichzeitig auszuführen (eines für die Position und eines für den Rest) und über eine Textdatei als Zwischenspeicher zu gehen. Das ist zwar kein wahres Multiprocessing, sondern geht eher in Richtung Multithreading. Einen Vorteil sollte es aber allemal bringen.