Denkschmied/fire-and-ice #40: Stromversorgung Raspberry-Pi über 5V GPIO und L298N H-Bridge

ampxtreme, 2019-05-29:
Servus
ich habe gerade das problem das mein Raspberry nicht starten kann mit der Stromversorgung über den Stihl-Akku (10,8V) --> H-Brücke L298N  (5V Jumer ist gesetzt) --> Raspberry Pi 5V GPIO.

Der RPi zeigt ein buntes Bild und versucht es dann wieder von vorne und sagt auch etwas von Stromversorgung...

Hatte jemand die Stromversorgung über denselben Pfad vor?

daholzmichl, 2019-05-30:
Servus,

ich hab 6 von den DC/DC Wandler bestellt: https://www.amazon.de/dp/B077VW4BTY/ref=pe_3044161_185740101_TE_item, werde die sobald geliefert testen und Feedback geben. Ich bin natürlich bereit einige davon auszuleihen falls notwendig / möglich.

LG
Alex

Denkschmied, 2019-06-03:
Klingt nach einer zu geringen Spannung...

ampxtreme, 2019-06-03:
Die haben wir gemessen. Das stellt eigentlich kein Problem dar. Jeder der die Möglichkeit hat noch eine Power-Bank zu nutzen sollte das tun, der Rest kann ggf. einen DC/DC-Wandler von Alex leihen!?

Vielleicht kann jeder kurz kommentieren welche Lösung er anstrebt, um ggf. noch Teile besorgen zu können:
- Amthor
- Gökmen
- Heigl
- Nocker
- Oberloher
- Schafferer
- Schröder
- Trocker
- Roth
- Sulzenbacher
- Aufinger
- Hagen --> PowerBank
- Maierhofer --> DC/DC

JoeUniv, 2019-06-05:
Ich kann leider noch immer nichts editieren, ich hab eine PowerBank.

ampxtreme, 2019-06-07:
@Denkschmied: Kannst Du bitte noch 3-4 PowerBanks mitbringen aus der FH, falls verfügbar? Danke!

closed 2019-06-07 by ampxtreme